redbear/nRF5x #7: Add ble nano 2 to mbed

jrobeson, 2017-03-24:
I see the BLE Nano 2 is not yet available as a target for mbed. I was able to get started by modifying the the PinNames.h, but it'd be nice to see it available officially.

jrobeson, 2017-05-08:
and I hope i did add it right, but I can't make printf() work at all, even though I set the pins to match the ble nano 2 pinout.

jrobeson, 2017-05-14:
i figured it out, it was the hardware flow control.

Srini2708, 2017-10-03:
Hi I too facing the same problem, My Printf() is not working, Can you suggest me the rootcause on how you configured the flow control... here is my code

include "mbed.h"
DigitalOut myled(P0_11);

Serial pc(P0_30, P0_29); // tx, rx

int main() {
    pc.baud(9600);

    while (true) {
        
       
        myled = 1;
        wait(1.5);
        myled = 0;
        wait(1.5);
        
        myled = 1;
        wait(0.2);
        myled = 0;
        wait(0.2);
        myled = 1;
        wait(0.2);
        myled = 0;
    
        pc.printf("Blink! LED is now ");
        wait(0.5);
    }
}

jrobeson, 2017-10-03:
use USBRX and USBTX  instead of those direct pin numbers.  I created/copied a patch that should contain all you need right here https://github.com/ARMmbed/mbed-os/pull/5231 . Just remove your current modifications and apply that.

jrobeson, 2017-11-14:
BLE Nano 2 support now merged into mbed-os: https://github.com/ARMmbed/mbed-os/pull/5231