redbear/nRF5x #11: Error during transfert

lc3t35, 2017-05-23:
Here is the full error log message : 
````
Arduino : 1.8.1 (Mac OS X), Carte : "BLE_Nano2"
*** cut ***
Using board 'BLE_Nano2' from platform in folder: /Users/laurent/Library/Arduino15/packages/RedBear/hardware/nRF52832/0.0.2
Using core 'RBL_nRF52832' from platform in folder: /Users/laurent/Library/Arduino15/packages/RedBear/hardware/nRF52832/0.0.2
Detecting libraries used...
"/Users/laurent/Library/Arduino15/packages/RedBear/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -mcpu=cortex-m4 -mthumb -mabi=aapcs -std=gnu++98 -Wall -Os -g -mfpu=fpv4-sp-d16 -ffunction-sections -fdata-sections -fno-strict-aliasing -fno-builtin 
**** cut ****
/T/arduino_build_796442/core/core.a" "/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442/core/mbed_os/platform/retarget.cpp.o"
Linking everything together...
"/Users/laurent/Library/Arduino15/packages/RedBear/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -Os -Wl,--gc-sections -mcpu=cortex-m4 -mthumb "-T/Users/laurent/Library/Arduino15/packages/RedBear/hardware/nRF52832/0.0.2/variants/BLE_Nano2/linker_scripts/gcc/BLE_Nano2.ld" --specs=nano.specs "-Wl,-Map,/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442/BLE_ScanReport.ino.map" -o "/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442/BLE_ScanReport.ino.elf" "-L/Users/laurent/Library/Arduino15/packages/RedBear/hardware/nRF52832/0.0.2/variants/BLE_Nano2/linker_scripts/gcc" "-L/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442" -lm -lgcc -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--warn-common -Wl,--warn-unresolved-symbols -Wl,--start-group "/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442/sketch/BLE_ScanReport.ino.cpp.o" "/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442/core/core.a" -lc -lnosys -Wl,--end-group
"/Users/laurent/Library/Arduino15/packages/RedBear/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-objcopy" -O binary "/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442/BLE_ScanReport.ino.elf" "/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442/BLE_ScanReport.ino.bin"
"/Users/laurent/Library/Arduino15/packages/RedBear/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-objcopy" -O ihex "/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442/BLE_ScanReport.ino.elf" "/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442/BLE_ScanReport.ino.hex"
Le croquis utilise 55648 octets (10%) de l'espace de stockage de programmes. Le maximum est de 524288 octets.
/Users/laurent/Library/Arduino15/packages/RedBear/tools/openocd/0.10.0-dev-nrf5/bin/openocd -d1 -s /Users/laurent/Library/Arduino15/packages/RedBear/tools/openocd/0.10.0-dev-nrf5/scripts/ -f /Users/laurent/Library/Arduino15/packages/RedBear/tools/openocd/0.10.0-dev-nrf5/scripts/interface/cmsis-dap.cfg -f /Users/laurent/Library/Arduino15/packages/RedBear/tools/openocd/0.10.0-dev-nrf5/scripts/target/nrf52.cfg -c init;reset halt;flash erase_address 0x7F000 0x1000;flash fillw 0x7F000 0x00000001 1;flash fillw 0x7F008 0x00000000 1;program {{/var/folders/mg/27j371qn68943s9ych8g50jm0000gp/T/arduino_build_796442/BLE_ScanReport.ino.bin}} verify reset exit 0x1C000 
Open On-Chip Debugger 0.10.0-dev-gdc53227 (2016-04-09-13:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
in procedure 'init' 
in procedure 'ocd_bouncer'

Une erreur est survenue lors du transfert du croquis
````

lc3t35, 2017-05-25:
I've used nRFToolbox on my iPad2 to verify that the BLE Nane2 was working fine : I can see the graph with HRM, I've noticed disconnection of device after ~75s, both app or device must be restarted to get bpm again.

lc3t35, 2017-05-25:
verifying configuration files : 
tools/openocd/0.10.0-dev-nrf5/scripts/interface/cmsis-dap.cfg 🆗 
tools/openocd/0.10.0-dev-nrf5/scripts/target/nrf52.cfg ok -> it seems to fail at "cortex_m reset_config sysresetreq" 
Let's try to increase debug level to 3

lc3t35, 2017-05-26:
Anybody from RedBear here ???

lc3t35, 2017-05-26:
Wrong side :(

closed 2017-05-26 by lc3t35

andreacassigoli, 2018-01-11:
I have the same problem with BLE Nano2.
If I drag&drop a precompiled hex file, everything works fine.
But if I try to upload (using Arduino IDE) a sketch, I receive this error:

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000007dc msp: 0x000008c0
erased address 0x0007f000 (length 4096) in 0.015026s (266.205 KiB/s)
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
wrote 4 bytes to 0x0007f000 in 0.640119s (0.006 KiB/s)
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
wrote 4 bytes to 0x0007f008 in 0.661060s (0.006 KiB/s)
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000007dc msp: 0x000008c0
** Programming Started **
auto erase enabled
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
wrote 20480 bytes from file C:\Users\Andrea\AppData\Local\Temp\arduino_build_664714/Blink_MIo.ino.bin in 1.252125s (15.973 KiB/s)
** Programming Finished **
** Verify Started **
verified 17996 bytes in 0.258091s (68.093 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

It seems to fails on "cortex_m reset_config sysresetreq".
Any help, please?

armaank, 2018-04-02:
Hi, 
I know that it has been a while, but I'm having the same issue. Were you able to find a fix to the problem? What's strange is that I was uploading via Arduino perfectly fine, but it just suddenly stopped working even after not making any changes. 

Thanks.

andreacassigoli, 2018-04-03:
Hi Armaan,

since it has been a while, you should explain me a little bit deeper what
the problem is, because I did not understand it.
Thank you.
Andrea

---------------------------------------

Andrea Cassigoli

Gestione Progetti Software

<http://www.easytechitalia.com/>

Via della Fangosa 32
50032 Borgo San Lorenzo (FI)
tel.055 8455216
www.easytechitalia.com

armaank, 2018-04-03:
Hi, 

The issue I have is the exact same as the issue that you were able to resolve here
http://discuss.redbear.cc/t/redbear-nano-v2-arduino-unable-to-upload/3060/16 
My error message is the same as the one given by user digsub. Except, whenever I attempt to follow your solution by dragging over the correct .bin file, the error persists when I attempt to burn the bootloader. 

Thanks

andreacassigoli, 2018-04-03:
> unable-to-upload/3060/16

Hi,

in my case I obviously followed exactly the procedure I posted in the forum.

The only thing could be the correct firmware version: even if in my Nano
board there was written version 1.5, the only formware file worked for me
was the " DAPLink_V1.0_Interface_nRF52.bin".

So, first try to upload different files from that folder.

then try again to upload your arduino sketch.

But remember also that the error message posted by digsub is not properly
an error message!!! It's just a warning/compiling message.
Even if it is red displayed, everything is well.
I always get this message after a sketch uploading, and everything works
fine.

let me know...

Andrea

---------------------------------------

Andrea Cassigoli

Gestione Progetti Software

<http://www.easytechitalia.com/>

Via della Fangosa 32
50032 Borgo San Lorenzo (FI)
tel.055 8455216
www.easytechitalia.com

armaank, 2018-04-04:
Hi,

Interesting. This is the exact message I get:
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
in procedure 'init' 
in procedure 'ocd_bouncer'

An error occurred while uploading the sketch

And the sketch does not upload. It is also worth mentioning that the chip itself seems to warm up? I'm not placing it into the DAPLINK incorrectly, so I'm not too sure about the cause. I'm using the v1.0 daplink, and I've been uploading the v1.0 bin to it. 

Thanks,

andreacassigoli, 2018-04-04:
When you upload the .bin file, you have only one chanche, because if the
.bin firmware file is not correct, you will not be able to run correctly
even the pre-compiled sketches.
So, after every .bin firmware file upload, please try to upload a
pre-compiled sketch (.hex) and verify that it is running correctly. If so,
that is the correctly .bin file for you.

After that, the message you wrote seems to be a real message error, and not
just a verbose debug log in uploading step.
Twice I placed the chip incorrectly on the DAPLink but no problems happened
and the chip continued to work. I never noticed that it warmed up.

About that error, I suppose you have to open a case to redbear, because
only the programmers could know what it means.

I hope to have been helpful to you.

Andrea

---------------------------------------

Andrea Cassigoli

Gestione Progetti Software

<http://www.easytechitalia.com/>

Via della Fangosa 32
50032 Borgo San Lorenzo (FI)
tel.055 8455216
www.easytechitalia.com