redbear/nRF5x #16: DFU and NVM

delcomp, 2017-07-03:
When do you expect to support DFU en the build-in NVM.

I'm near the release of our new product using the BLE Nano 2. But without DFU or NVM we can't release.

Rudi

Cheong2K, 2017-07-04:
Nordic's SDK already includes the DFU, you can use it. What do you mean by NVM?

delcomp, 2017-07-04:
DFU is indeed supported by Nordic's SDK. But it does not work with the bootloader you made for the RF52832. The nRFToolBox (iOS) DFU finds the BLE Nano 2, but can not do the firmware upgrade. I've also added the DFU services to my Arduino sketch. I notice that when the nRFToolBox DFU app starts the BLE Nano 2 does a reset, enters the DFU mode but can not receive the new firmware hex file. I thought this issue was known.
I should mention that the same Arduino sketch works perfectly with a BLE Nano 1.5

Note: I've patched DUFService.h/cpp to remove the #ifdef TARGET_NRF51822 to test the DFU in my BLE Nano 2 Arduino sketch.

NVM = non volatile memory. The nRF52832 has some NVM integrated. It can be used to store for example user configurations, serial numbers, ... without the need for an external flash or eeprom. You could support it in Arduino and include an example sketch.

Cheong2K, 2017-07-05:
The source and binary, we modified from Nordic's SDK:
https://github.com/redbear/nRF5x/tree/master/nRF52832/dfu

We tested and it works, please follow this:
https://github.com/redbear/nRF5x/blob/master/nRF52832/docs/DFU_User_Guide.md

delcomp, 2017-07-05:
I'm sorry but it does not work. May I ask you to retest this and use an iPhone (iOS) and the nRF Toolbox App. I ask this because I've read on some other forum that the Android version might work, but the iOS does not, I can not verify this cause I do not have an Android device.

Regarding your DFU_User_Guide.md:
Starting DFU mode works fine, the leds start blinking at 100msec. But the nRF Toolbox App then does not seem to be able to transmit the new program. Finally the module times out after 30 seconds (as in your guide mentioned).

Also not that point 2 in your Guide 'Add fu service to application' is not possible because in your DFUService.h/cpp files there is line which removes the code for anything else but a NRF51822. As I've mentioned before I've commented this line, then the DFU service can be included in my app. This parts works fine.

Cheong2K, 2017-07-11:
The DFU guide is updated, would you mind to test it again? Thanks for your information, too.
https://github.com/redbear/nRF5x/blob/master/nRF52832/docs/DFU_User_Guide.md

delcomp, 2017-07-11:
Read your user guide. So I should conclude that .hex files can not be used to upgrade the BLE Nano 2. The hex files must first be converted to zip files. Correct ?

I'm using an iMac, not a Windows PC. The instructions to create a zip file in your guide do not work. I installed python, pip and nrfutil using the instructions found in: https://github.com/NordicSemiconductor/pc-nrfutil

The command line command nrfutil works, but the dfu command 'genpkg' is a unknown command. How can I add this command, or better please include detailed and correct instructions on how to create the required zip file on an iMac.

Further more I did something foolish. My test BLE Nano module could be put in DFU mode by putting P0_04 to GND, the led was blinking when I did that. I did not know whether the correct booloader/softdevice was installed on my BLE Nano 2 (how to know this???). So I dragged the  SoftdevicesS132_2.0_SDK11_Bootloader_20161221.hex file to the DAPLINK drive. The BLE Nano 2 reset and I presume installed the hex file because not FAIL.TXT was created. Only after doing this I can not get the led blinking again by putting P0_04 to GND.
So 2 questions:
- how to read the bootloader version number
- how to upgrade the softdevice/bootloader

delcomp, 2017-07-11:
OK, upgraded the DAPLINK with the firmware DAPLINK_V1.5_Interface_nRF52_20170106.bin. Then tried again with copying SoftDevicesS132_2.0_SDK11_Bootloader_20161221.hex to the DAPLINK drive. After this the BLE Nano 2 softdevice/bootloader was done correctly and the led starts blinking again when P0_04 is grounded.

I also was able to generate a zip file using the commands found on above mentioned website:
nrfutil keys generate private.pem
nrfutil pkg generate --hw-version 52 --sd-req 0x81 --application-version 0xFFFFFFFF --application BLE_Track.ino.hex --key-file private.pem BLE_Track.zip

This resulted in a zip file: BLE_Track.zip.
I copied this to the nRF Toolbox app and tried again the DFU upgrade.

Unfortunately with the same result as with the hex file.
- led blinks on the BLE Nano 2
- DFU app works fine in selecting BLE Nano 2 and selecting file. But it gets stuck at Upload. I mean 'Upload' is shown under 'DEVICE FIRMWARE UPGRADE'. This stays for a very long time then times out and the upgrade was not done.

My conclusion after 4 hours work: neither HEX or ZIP files work with the DFU procedure.

delcomp, 2017-07-11:
Got a Windows machine and installed the Master Control Panel (please add a link to this in your user_guide cause I couldn't easily find it). 
Then created the zip file per your user_guide. And Eureka, the DFU does work.
For an OSX user it's not a nice solution when a Windows PC is needed just for creating a ZIP file.

So if you could arrange this ZIP can be created on a OSX machine, that would be the final solution.

pentabarf, 2018-02-02:
@delcomp, i had the same issue. the solution was to downgrade nrfutil to 0.5.2
https://github.com/NordicSemiconductor/pc-nrfutil/tree/0_5_2
on github they mention to use this version for the old unsecured bootloader!

how did you get DFUService on Arduino program for BLE Nano2 working? Just removing #ifdef TARGET_NRF51822 in DFUService.h/cpp?
Can you please post your Code Lines in Main program for activating DFUService?!