redbear/nRF5x #24: Pairing difficulties (Nano V2 - BLE_Serial - Arduino IDE)

LaurentCabaret, 2017-11-11:
I experience strong pairing difficulties with the red bear Nano v2.0.
I correctly uploaded the BLE_Serial example and the “TXRX” appears on some of my computers (only recent ones).

But when I try to pair the devices (nano2 with  computer) there is only time-out.

Any idea ?

Thanks in advance,
Laurent

Arudinne, 2018-02-14:
I was having a similar issue. The "TXRX" device would show up but when I tried to connect to it with various UART apps it would drop the connection saying there was no device with the UART service or something to that effect.

When trying to connect with the Nordic UART app it kept reporting it as "NORDIC_HRM" even though the sketch successfully uploaded and other non-BLE sketchess worked fine.

It looks like RedBear has the wrong UUID configured in the BLE_Serial sketch. I reconfigured the UUIDs per this page and it worked: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk52.v0.9.2%2Fble_sdk_app_nus_eval.html

pucchakayalas, 2018-06-20:
@Arudinne I changed the UUIDS from my Arduino IDE and now it shows the "Nordic UART Service" in the nordic connect application, but still it is not getting bonded. Please let me know how to resolve this problem.

@LaurentCabaret Any luck?

Arudinne, 2018-06-20:
I used the UUIDs mentioned on that page and it worked for me during some brief testing. If that didn't fix your issue then I am unfortunately not able to help you as I am still fairly new to embedded programming.

I've pretty much given up on these devices because RedBear has effectively abandoned their entire product line mere months after I bought my nano v2s. The announcement is on their page: https://redbear.cc/ 

Had I known this was going to happen I wouldn't have wasted my money.

My recommendation: Buy something else.

pucchakayalas, 2018-06-22:
@Arudinne Thank you for the reply, Looks like I got a bad deal. I am currently using it for my university project and I hope I get the job done.