redbear/nRF5x #26: DAPLink firmware issues with UART/USB on BLE Nano 2

rsalveti, 2017-12-12:
I'm trying to use BLE Nano 2 as a bluetooth LE dongle powered by Zephyr (e.g. https://devzone.nordicsemi.com/blogs/1059/nrf5x-support-within-the-zephyr-project-rtos/), but had several UART and USB issues with the stock firmware.

As I noticed there is already a pull request for upstream DAPLink support at https://github.com/ARMmbed/DAPLink/pull/178, I took it as reference and updated to the latest DAPLink head, including updates to the UART (circular buffer and flow control) and USB (reduced interval for CDC). You can check my latest branch at https://github.com/OpenSourceFoundries/DAPLink/tree/blenano2-osf.

You can also find the pre-built firmware (from the blenano2-osf branch) at http://rsalveti.net/tmp/blenano2/DAPLink_V1.5_Interface_nRF52_20171212.bin, in case you want to give that a try.

anderspitman, 2018-01-27:
Hey @rsalveti, what exactly is the scope of the DAPLink issues I can expect to have until this is fixed? Currently programming the Nano 2 via pyocd seems to be working fine for me. Haven't been able to get debugging up and running yet though. What can I expect to work/not work?

rsalveti, 2018-01-30:
Mostly USB/UART related, it can get stuck if you stress the interface.

Debugging should still work fine with the factory firmware, try using pyocd-gdbserver and gdb (connect via telnet).

anderspitman, 2018-04-26:
@rsalveti thanks for the tip. I've got gdbserver up and running, and I'm able to connect, but I'm not seeing any of my `printk` output showing up. I tried running `continue` from within gdb. Sorry I know this is a super basic problem but I'm new to debugging with zephyr.

anderspitman, 2018-04-26:
nvm [figured it out](https://github.com/zephyrproject-rtos/zephyr/issues/7219)! Thanks again for the tip though!

ericlangel, 2019-03-07:
Is there a Redbear DapLink V1.5 firmware with flowcontrol on RTS/CTS?
Or with a DTR Pin?