redbear/nRF5x #29: High Power Consumption

unCleanCode, 2017-12-28:
I'm trying to lower down power consumption in advertising mode down to 100-200uA, but all the time getting around 1.2mA !

Even in deep sleep mode (SYSTEMOFF) when expected power consumption should be about 6uA I'm getting crazy 350uA !

I'm powering BLE Nano 2 with single CR2032 coin battery. I tried powering it through VDD and VIN, result is the same

Code I'm using for advertising mode (not DEEP SLEEP):
`#include <nRF5x_BLE_API.h>

#define DEVICE_NAME            "BLE_Debug"
#define TXRX_BUF_LEN           20

BLE                            ble;

static const uint8_t service1_uuid[]        = {0x71, 0x3D, 0, 0, 0x50, 0x3E, 0x4C, 0x75, 0xBA, 0x94, 0x31, 0x48, 0xF1, 0x8D, 0x94, 0x1E};
static const uint8_t service1_tx_uuid[]     = {0x71, 0x3D, 0, 3, 0x50, 0x3E, 0x4C, 0x75, 0xBA, 0x94, 0x31, 0x48, 0xF1, 0x8D, 0x94, 0x1E};
static const uint8_t service1_rx_uuid[]     = {0x71, 0x3D, 0, 2, 0x50, 0x3E, 0x4C, 0x75, 0xBA, 0x94, 0x31, 0x48, 0xF1, 0x8D, 0x94, 0x1E};
static const uint8_t uart_base_uuid_rev[]   = {0x1E, 0x94, 0x8D, 0xF1, 0x48, 0x31, 0x94, 0xBA, 0x75, 0x4C, 0x3E, 0x50, 0, 0, 0x3D, 0x71};

uint8_t tx_value[TXRX_BUF_LEN] = {0,};
uint8_t rx_value[TXRX_BUF_LEN] = {0,};

GattCharacteristic  characteristic1(service1_tx_uuid, tx_value, 1, TXRX_BUF_LEN, GattCharacteristic::BLE_GATT_CHAR_PROPERTIES_WRITE | GattCharacteristic::BLE_GATT_CHAR_PROPERTIES_WRITE_WITHOUT_RESPONSE );
GattCharacteristic  characteristic2(service1_rx_uuid, rx_value, 1, TXRX_BUF_LEN, GattCharacteristic::BLE_GATT_CHAR_PROPERTIES_NOTIFY);
GattCharacteristic *uartChars[] = {&characteristic1, &characteristic2};
GattService         uartService(service1_uuid, uartChars, sizeof(uartChars) / sizeof(GattCharacteristic *));

void disconnectionCallBack(const Gap::DisconnectionCallbackParams_t *params) {
  ble.startAdvertising();
}

void gattServerWriteCallBack(const GattWriteCallbackParams *Handler) {

}

void setup() {

  NRF_POWER->DCDCEN = 0x00000001;
  NRF_UART0->TASKS_STOPTX = 1;
  NRF_UART0->TASKS_STOPRX = 1;
  NRF_UART0->ENABLE = 0;
  pinMode(13, INPUT);
  ble.init();
  ble.onDisconnection(disconnectionCallBack);
  ble.onDataWritten(gattServerWriteCallBack);

  ble.accumulateAdvertisingPayload(GapAdvertisingData::BREDR_NOT_SUPPORTED);
  ble.accumulateAdvertisingPayload(GapAdvertisingData::SHORTENED_LOCAL_NAME,
                                   (const uint8_t *)"TXRX", sizeof("TXRX") - 1);
  ble.accumulateAdvertisingPayload(GapAdvertisingData::COMPLETE_LIST_128BIT_SERVICE_IDS,
                                   (const uint8_t *)uart_base_uuid_rev, sizeof(uart_base_uuid_rev));
  ble.setAdvertisingType(GapAdvertisingParams::ADV_CONNECTABLE_UNDIRECTED);
  ble.addService(uartService);
  ble.setDeviceName((const uint8_t *)DEVICE_NAME);
  ble.setTxPower(4);
  ble.setAdvertisingInterval(1000);
  ble.setAdvertisingTimeout(0);
  ble.startAdvertising();
}

void loop() {
    ble.waitForEvent();
}`

unCleanCode, 2017-12-28:
Wow, that's strange. I bought two of Nano 2 and just put the same code on second one and it consumes below 100uA!

What can be the problem with the first one? SoftDevice? Can I checj version of it?

colin-guyon, 2017-12-29:
350 uA make me think of some current flowing through a pin... (I believe I had the same measurement with a input pin being connected to ground and configured with pull up resistor)...are you sure the circuitry is the same for both ble nano ?

unCleanCode, 2017-12-29:
No, circuit is the same (in fact for test purposes all I have in my circuit is battery, multimeter and BLE Nano 2. No resistors, no extra wires, no LEDs, nothing. So eithere there is some factory bug (which I probably won't see as MB-N2 has metal cover/radiator), or nRF52 device got into some strange mode which consume more current (at start I thought it's DFU, but in few minutes I understood that it doesn't behave like DFU. Now I'm thinking on nRF52 Debug mode, but code I found [here ](https://devzone.nordicsemi.com/question/164786/exit-from-debug-mode/): 
`DWT->CYCCNT != 0` shows me I'm not

What other state can be saved on board which isn't flushed with new program code?

unCleanCode, 2017-12-29:
Btw, I simplified my code to the following:

```
#include <nRF5x_BLE_API.h>

void setup() {
  NRF_POWER->SYSTEMOFF = 1;
}

void loop() {}
```

This code on my normal Nano 2 gives 4uA consumption. On the second one it shows around 300uA

colin-guyon, 2017-12-29:
ok very strange indeed... Hope someone will be able to help

giowild, 2018-03-31:
Hi @unCleanCode, I would like to measure current consumption as you are doing. Can you detail better which kind of equipment you are using, how do you connect it to the Nano and how much is it?

Thanks in advance!!!

unCleanCode, 2018-04-05:
Hi @giowild ,

equipement - multimeter. Recently updated mine to Amprobe AM-520 HVAC if that's important

connection - simply putting it in series connection. Sorry, not sure is that the right term in english. So I connect it in a way that the current goes from the battery to one probe of the multimeter, through the multimeter to another probe and through it to the VIN pin. Ground pin of the board is connected to the other pole of the battery. All the current consumption results are written above ... not sure if you need anything else from that

closed 2025-07-01 by unCleanCode