redbear/nRF5x #31: Question : Maximum BAUD rate for sending data over bluetooth

thedub2001, 2017-12-29:
I can't find any information on the web about bluetooth Tx/Rx speed of the BLE Nano 2 module.
Where and how do I set it up ?
What are the available speeds ?
Thanks a lot

jcbooth2, 2018-04-14:
I would not touch that unless you know what you are doing. That value is used internally and might mess up everything. There are other ways to transmit data faster. (Assuming that is what you are trying to do)

Bluetooth Low Energy is a little different than Bluetooth Classic. Bluetooth classic with the SPP profile (what most hobbyists are used to) emulates a classic RS232 with a constant (or nearly constant) stream of data being transmitted back and forth. It is much different with Bluetooth Low Energy where data is sent in 20 byte packets every connection interval. You can emulate an Rx and Tx buffer through BLE but it is not really the same thing. What is actually happening is on your peripheral you are creating a UART Service with a Rx and Tx characteristics. When you update the Tx buffer with a new value, and the client device requests the new information, then the data placed into the Tx buffer is split into 20 byte packets and transmitted across as many connection intervals as it takes to send the data.

The idea of a connection interval is what saves so much energy and what slows down the transmission rate. I have had to fight it significantly in order to get the data transmission that I want. I was able to avoid the throughput limit by writing my own custom service and characteristics.

[Here](https://learn.adafruit.com/introduction-to-bluetooth-low-energy/introduction) is a more in depth explanation of Bluetooth Low Energy.

[Here](https://www.novelbits.io/bluetooth-5-speed-maximum-throughput/) is a good article about how achieve a high throughput with low latency using the BLE, but it gets very technical.

[Here](https://punchthrough.com/blog/posts/maximizing-ble-throughput-on-ios-and-android) is another article that is less technical.

I hope this helps! The nRF52832 is a great chip!

PiedPieper, 2019-02-19:
@jcbooth2 Did you experience issues with transmitting Serial to BT ?

Right now I have my serial line set to a baud rate of 9600 and the nRF is dropping a significant amount of data. It seems that Serial.attach(callback_function) is the culprit, and that interrupts aren't firing. Did you ever solve this issue?

I have found that if I remove Serial.attach(), don't enable BLE, and just sit on a : 
```
While(1) {
    while(Serial.available()) {
        Serial.read()....
    }
}
```

Then it reads far faster, but it still drops data.