redbear/nRF5x #50: Hardware without XTAL and without DCDC

ericlangel, 2018-10-10:
Hello,

i have a NRF52832 Module without 32.768kHz X'tal and without DCDC.
But the Redbear examples doesn't work at all.
The Nordic Support said there are config headers which have to be changed to the right clock source and power source.

So which files have to be changed to use internal LDO and RC oscillator?

i found the "device.h" file and changed the Clock_Src to 4, but this doesn't help.
/**
 * @brief NRF_LF_SRC_XTAL  2
 *        NRF_LF_SRC_SYNTH 3
 *        NRF_LF_SRC_RC    4
 */
#define MBED_CONF_NORDIC_NRF_LF_CLOCK_SRC 4

where i can find the Power settings?

I hope someone can help me

ericlangel, 2018-11-15:
Hello mtiutiu,

thank you for your reply. I will test it the next days.
After posting this issue i got a possible work around from Nordic.

in the "device.h" in ...\RedBear\hardware\nRF52832\0.0.2\cores\RBL_nRF52832\mbed_os\targets\TARGET_NORDIC\TARGET_NRF5\TARGET_MCU_NRF52832\TARGET_NRF52_DK

there is the definition for the LF Clock

/**
 * @brief NRF_LF_SRC_XTAL  2
 *        NRF_LF_SRC_SYNTH 3 
 *        NRF_LF_SRC_RC    4
 */
#define MBED_CONF_NORDIC_NRF_LF_CLOCK_SRC 2
#define MBED_CONF_NORDIC_UART_HWFC 0
#define MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE 9600

with the SYNTH Clock selected i got my Application running. But not with the RC Clock.
And i had to use the original Softdevice 2.0.0 from Nordic.
With the Redbear Softdevice it didnt work.

mtiutiu, 2018-11-15:
Great. Well in my case I used a **PTR5518** module which is **nrf51822** based and without the LF XTAL mounted. It didn't worked with the redbear standard bootloader obviously but after modifying the bootloader it started to work like a charm without doing other changes related to this Arduino nRF5x core. 

I also created a [PR](https://github.com/redbear/nRF5x/pull/52) for this one but I don't think it will get approved because I don't think it will be of interest for @ChiHung or RedBear company from obvious reasons.